iNavFlight/blackbox-log-viewer #13: Unable to open log file

preisedj, 2018-01-17:
I downloaded the latest version and tried to open a file.   Nothing happens, there is no message.  I've tried several other log files.
I can open the file with the Cleanflight/Blackbox-log-viewer just fine.

DzikuVx, 2018-01-18:
@preisedj please provide a log that fails to open, I can not confirm this with my logs

xermalk, 2018-02-28:
This log fails to open in INAV Blackbox Explorer 3.0.1. it is a logfile capured over usb on INAV/OMNIBUSF4PRO 1.8.0.

"Error opening log: No logs in this file could be parsed successfully"

Do i need a older blackbox viewer for it to work? 

`timestamp,gyroscopeX,gyroscopeY,gyroscopeZ,accelerometerX,accelerometerY,accelerometerZ,magnetometerX,magnetometerY,magnetometerZ,kinematicsX,kinematicsY,kinematicsZ,altitude,gpsFix,gpsNumSat,gpsLat,gpsLon,gpsAlt,gpsSpeed,gpsGroundCourse,voltage,amperage,mAhdrawn,rssi,RC0,RC1,RC2,RC3,RC4,RC5,RC6,RC7,RC8,RC9,RC10,RC11,RC12,RC13,RC14,RC15,RC16,Motor0,Motor1,Motor2,Motor3,Motor4,Motor5,Motor6,Motor7,Debug0,Debug1,Debug2,Debug3
1519844087028,-1.2195121951219514,0.24390243902439027,0.48780487804878053,-0.0703125,0.119140625,0.978515625,0,0,0,6.9,4.1,359,-0.12,2,6,56.2011934,12.5739311,10,8,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844088027,0.48780487804878053,-0.48780487804878053,0,-0.0703125,0.1171875,0.9765625,0,0,0,6.9,4.1,359,-0.09,2,6,56.2011933,12.5739307,10,16,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
timestamp,Debug0,Debug1,Debug2,Debug3
1519844095326,0,0,0,0
1519844096327,0,0,0,0
1519844097327,0,0,0,0
1519844098326,0,0,0,0
timestamp,gyroscopeX,gyroscopeY,gyroscopeZ,accelerometerX,accelerometerY,accelerometerZ,magnetometerX,magnetometerY,magnetometerZ,kinematicsX,kinematicsY,kinematicsZ,altitude,gpsFix,gpsNumSat,gpsLat,gpsLon,gpsAlt,gpsSpeed,gpsGroundCourse,voltage,amperage,mAhdrawn,rssi,RC0,RC1,RC2,RC3,RC4,RC5,RC6,RC7,RC8,RC9,RC10,RC11,RC12,RC13,RC14,RC15,RC16,Motor0,Motor1,Motor2,Motor3,Motor4,Motor5,Motor6,Motor7,Debug0,Debug1,Debug2,Debug3
1519844104371,0.48780487804878053,0.24390243902439027,0,-0.0703125,0.115234375,0.978515625,0,0,0,6.8,4.1,359,-0.12,2,6,56.2011511,12.5739414,13,8,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844105372,1.2195121951219514,-0.24390243902439027,0,-0.0703125,0.1171875,0.9765625,0,0,0,6.9,4.1,358,-0.13,2,6,56.2011506,12.5739438,12,6,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844106371,0.48780487804878053,0,0.24390243902439027,-0.068359375,0.119140625,0.98046875,0,0,0,6.9,4.1,358,-0.18,2,6,56.2011522,12.5739472,11,22,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844107372,1.9512195121951221,0,0.48780487804878053,-0.0703125,0.1171875,0.98046875,0,0,0,6.8,4.1,358,-0.25,2,6,56.2011506,12.5739488,11,11,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844108371,-1.707317073170732,-0.24390243902439027,0.48780487804878053,-0.068359375,0.12109375,0.978515625,0,0,0,6.8,4.1,358,-0.3,2,6,56.2011479,12.5739504,12,10,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844109371,0.9756097560975611,-0.24390243902439027,0.24390243902439027,-0.0703125,0.11328125,0.96875,0,0,0,6.8,4.1,358,-0.31,2,6,56.2011441,12.573951,12,9,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844110371,-2.439024390243903,0,0.48780487804878053,-0.0703125,0.12109375,0.974609375,0,0,0,6.9,4.1,358,-0.28,2,6,56.201143,12.5739531,12,2,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844111371,0.48780487804878053,-0.24390243902439027,0.24390243902439027,-0.072265625,0.119140625,0.98046875,0,0,0,6.9,4.1,358,-0.25,2,6,56.2011421,12.5739542,12,11,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844112371,-1.2195121951219514,0,0.48780487804878053,-0.0703125,0.119140625,0.9765625,0,0,0,6.8,4.1,358,-0.23,2,6,56.2011395,12.5739535,12,11,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
timestamp,gyroscopeX,gyroscopeY,gyroscopeZ,accelerometerX,accelerometerY,accelerometerZ,magnetometerX,magnetometerY,magnetometerZ,kinematicsX,kinematicsY,kinematicsZ,altitude,gpsFix,gpsNumSat,gpsLat,gpsLon,gpsAlt,gpsSpeed,gpsGroundCourse,voltage,amperage,mAhdrawn,rssi,RC0,RC1,RC2,RC3,RC4,RC5,RC6,RC7,RC8,RC9,RC10,RC11,RC12,RC13,RC14,RC15,RC16,Motor0,Motor1,Motor2,Motor3,Motor4,Motor5,Motor6,Motor7,Debug0,Debug1,Debug2,Debug3
1519844117764,1.9512195121951221,-0.24390243902439027,0,-0.0703125,0.1171875,0.978515625,0,0,0,6.9,4.1,358,-0.07,2,6,56.2011378,12.5739543,12,8,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844117814,1.9512195121951221,-0.24390243902439027,0,-0.0703125,0.1171875,0.978515625,0,0,0,6.9,4.1,358,-0.07,2,6,56.2011327,12.5739649,12,7,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844117864,-0.7317073170731708,-0.48780487804878053,0.48780487804878053,-0.07421875,0.1171875,0.98046875,0,0,0,6.9,4.1,358,-0.07,2,6,56.2011327,12.5739649,12,7,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0 [... cut ...]
1519844129665,-4.634146341463415,0.7317073170731708,0.48780487804878053,-0.072265625,0.103515625,0.966796875,0,0,0,6.3,4.1,357,0.1,2,6,56.2011248,12.573962,15,17,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
1519844129715,4.146341463414634,-0.7317073170731708,0,-0.072265625,0.109375,0.98046875,0,0,0,6.3,4.1,357,0.1,2,6,56.201125,12.5739622,15,5,2878,0,0,0,0,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1500,1000,0,0,0,0,0,0,0,0,0,0,0
`

DzikuVx, 2018-03-01:
This log is truncated, does not contain header and just is not a valid BB log

closed 2018-05-12 by DzikuVx