iNavFlight/blackbox-log-viewer #25: INAV blackbox-log-viewer runs but will not open log files

steviegeek, 2018-12-28:
INAV blackbox-log-viewer downloaded today 28 Dec 2018 starts in Chrome browser, allows me to select a log file to open but does not open it, does nothing !, no error messages etc..
Log files tried are 2,857KB and 10,096KB recorded yesterday on SDcard using MATEK F722 STD running v2.0.1
CHROME says I'm running latest version, Version 71.0.3578.98 (Official Build) (64-bit), on Win7 PRO on Intel i5
Assistance most welcome :-)
Thanks Steve

[image]

steviegeek, 2018-12-28:
BTW, these log files open OK in Cleanflight and Betaflight Blackbox Explorers
Sreve

steviegeek, 2018-12-28:
Was using old file, found latest, all OK

closed 2018-12-28 by steviegeek

i3130002, 2019-08-22:
May I ask for some of your sample log files ? I want to convert my logs into INVA ones so I need some working logs. Thanks