iNavFlight/blackbox-log-viewer #32: save Workspace is a json filled with nulls

laurentopia, 2019-03-03:
Is this function implemented? If so, what is its purpose?
Before I close a BB log session I save the workspace, hoping that it saves log to video offset but the file generates is a null json

> {
    "graphConfig": [
        null,
        null,
        null,
        null,
        null,
        null,
        null,
        null,
        null,
        null
    ]
}

Tonynux, 2024-03-26:
And after 5 years we are at the same point!

DzikuVx, 2024-03-27:
yes @Tonynux you are more than welcome to work on this problem just like any other open source user.

Tonynux, 2024-03-27:
@DzikuVx yes, I understand your point of you, and of course you are right! I would like to be able to solve this problem, but unfortunately I am not.
Pawel, may you please just paste here some workspace.json file template, so anyone can manually customize the file to be imported il Log-Viewer?!

druckgott, 2024-04-09:
Fixed it:
https://github.com/iNavFlight/blackbox-log-viewer/pull/99