iNavFlight/blackbox-log-viewer #40: "Error opening log"

VitDetz, 2019-10-30:
Blackbox problem: "Error opening log: No logs in this file could be parsed successfully"
INAV Blackbox Explorer version v3.3.0 
FC - Matek F405ctr
Changing the recording speed modes: 100/88/80/75/67/50 /% ... etc., as well as replacing the SD card and reformatting it, did not solve the problem.
Replacing the flight controller with a similar one did not solve the problem.
Plasmatree / PID-Analyzer - crash when trying to open a log file.
Your cakes are inedible gentlemen! It's time to start cooking normal food)))
[LOG00001.TXT](https://github.com/iNavFlight/blackbox-log-viewer/files/3787332/LOG00001.TXT)

AlexAlle2020, 2020-06-05:
and also OMNIBUS F4 NANO V6 / Error opening log: No logs in this file could be parsed successfully

DzikuVx, 2021-10-31:
Log is damaged

closed 2021-10-31 by DzikuVx