iNavFlight/blackbox-log-viewer #43: Altitude should be on the same Y scale (i.e. navPos[2](m) and baro (m))

mf0o, 2020-06-17:
**current state:**
Debugging altitude issues is quite hard as the navPos[2] seems to be on a different Y-scale as the baro(m).

Here for example the readings are almost identical, but the lines are not close to eachother
[image]

A few seconds later, barometer reads 56m, so I would expect the green line higher than the red navPos[2].
[image]

---
**expected behaviour:**

It should more look like this (data exported and graphed), the red bar marks the same second/altitude as in the BBE above:
[image]

---
**version:**
- iNav 2.4
- BBE 3.4.0

spatzengr, 2023-10-25:
fixed in #87

spatzengr, 2023-10-25:
@pawel-spychalski, you should be able to close this issue too.  :-)

closed 2023-10-25 by DzikuVx