iNavFlight/blackbox-log-viewer #49: Cannot play any video on Linux

skorokithakis, 2020-12-19:
When trying to load any video, I get "could not play this video, try Chrome". While it's odd that a desktop app asks me to try Chrome, I think this is actually not a video issue (it happens with many formats), but a path issue, because of the following steps:

1. I click "open video".
2. I select "/tmp/myvideo.mp4".
3. "Video could not be loaded, try Chrome".
4. I click "open video" again.
5. The dialog box says "_tmp_myvideo.mp4".

I think the explorer tries to sanitize the path and then can't find the video at the sanitized location, so it fails.

skorokithakis, 2020-12-19:
Just to mention, I have recoded the video as per the README to no avail.

closed 2022-07-07 by DzikuVx

xznhj8129, 2024-11-25:
Yeah this is still a problem, what do i do? I don't use windows