iNavFlight/blackbox-log-viewer #57: Error opening log

robustini, 2021-06-13:
Flown today, log not readable.
Too bad because there was a problem to analyze, a return to home where there was an anomalous behavior of the throttle. 

[blackbox_log_2021-06-13_183730.zip](https://github.com/iNavFlight/blackbox-log-viewer/files/6644425/blackbox_log_2021-06-13_183730.zip)

aLDime, 2021-07-16:
`Flown today, log not readable.`
The log is also not readable ... 
There were a lot of problems with RTH, how to look?
[blackbox_log_2021-07-16_164510.zip](https://github.com/iNavFlight/blackbox-log-viewer/files/6831254/blackbox_log_2021-07-16_164510.zip)
[INAV_cli_Mark4_7inch_20210716_164358(Bat_Cur-mAh_Calib).zip](https://github.com/iNavFlight/blackbox-log-viewer/files/6831299/INAV_cli_Mark4_7inch_20210716_164358.Bat_Cur-mAh_Calib.zip)

DzikuVx, 2021-10-31:
Log was opened in the latest version no problem 
[image]

closed 2021-10-31 by DzikuVx