iNavFlight/blackbox-log-viewer #70: Can’t launch the app on macOS BigSur

roroid, 2021-12-21:
I was trying to launch the latest release of the app on macOS BigSur. Unfortunately, the app is not opening. I can see the app menu in the top bar but when trying to open a file app closes. I was opening the app considering that it was not signed.

z2cooleo, 2021-12-27:
On macOS Monterey has the same issue.

vinc13008, 2022-01-02:
+1 Same on Monterey

roroid, 2022-01-03:
I have to build my executable from the source code, which somehow works. If someone else needs it, I can post the link.

Sjullep, 2022-04-07:
+1 Same on Monterey

febbyeka, 2022-04-30:
+1 same on my Monterey

Pairan, 2022-05-14:
> I have to build my executable from the source code, which somehow works. If someone else needs it, I can post the link.

that would be nice - thanks!

erstec, 2022-06-06:
Let's make build different way
Please try
https://www.icloud.com/iclouddrive/08cA-CcEEwR-rq-dQGPhlEPMQ#INAV-BlackboxExplorer%5FmacOS%5F4.0.0

erstec, 2022-06-09:
NW updated to 0.6.1, if someone can test it - it would be nice.
https://www.icloud.com/iclouddrive/0c5_lzYpYso0S2pH8VEqxH8xQ#INAV-BlackboxExplorer_macOS_4.0.0_2

martinmichaelfuchs, 2022-07-05:
Same problem here. I installed the INAV Blackbox Explorer 4.0 on macOS Big Sur version 11.6 but the Blackbox Explorer does not open. I know that the app isn't signed and I allowed my Mac to run it anyways, so that's not the issue. I see the app name in the top bar but can't get anywhere else

Pairan, 2022-07-05:
Download it and it WORKS AS EXPECTED @erstec  Thank you so much for letting uns dig into our logs again!

Bryksin, 2022-07-05:
The same, previous version does work, latest 4.0 does not open! UI not appearing

closed 2022-07-06 by DzikuVx