iNavFlight/blackbox-log-viewer #91: Frequency axis on Analyzer view shows infinite frequency

aviphysics, 2023-12-16:
The frequency axis in the analyzer view shows infinite frequency. I am using the latest release version of iNav7 and the Blackbox log viewer available at the time of posting. The data from iNav appears valid and I can do the FFT and plot in Octabe, so this seems to be a problem with the blackbox log viewer.

[image]
[blackbox_log_2023-12-15_123126.TXT](https://github.com/iNavFlight/blackbox-log-viewer/files/13690863/blackbox_log_2023-12-15_123126.TXT)

Phil-MC, 2023-12-17:
I've seen a couple of comments on the forums with users asking what the frequencies are. The axis label seems to have taken over from the scale labels in the display.

aviphysics, 2023-12-17:
> I've seen a couple of comments on the forums with users asking what the frequencies are. The axis label seems to have taken over from the scale labels in the display. I can't find the word infinite in the code. @DzikuVx

Did you try looking for "infinity"?

Phil-MC, 2023-12-18:
I did look for infinity as well using github search and there were no hits in the code. I hope @DzikuVx has a look at this problem. I can see lots of users having problems with it. The graphs look OK, it's just the labelling.

rts18, 2024-01-14:
@spatzengr If you find some time. Can you please take a look at this issue. Possibly caused by #86

spatzengr, 2024-01-14:
@rts18 , yes.  I noticed the same issue but didn't associate it with #86 , but I can investigate to see wsup.

balbusky21, 2024-03-05:
I'm having the same issue. Is there a fix yet for this? I'm having issues tunning my filters. or is there a work around?

aviphysics, 2024-03-05:
> I'm having the same issue. Is there a fix yet for this? I'm having issues tunning my filters. or is there a work around?

My current work around is to use the last version of BBV

Phil-MC, 2024-03-06:
I gather it can be caused by missing entries in the BB Recording. Try reducing the amount of data collected. Some of the FC's are at the limit of what their design can handle.

aviphysics, 2024-03-06:
> I gather it can be caused by missing entries in the BB Recording. Try reducing the amount of data collected. Some of the FC's are at the limit of what their design can handle.

I don't think that is the problem here (or not the entire problem)

For one thing, BBV5 can render the FFT with scale without a problem from the same dataset that doesn't work in BBV7.

Have you been able to demonstrate a working FFT scale in BBV7?

Phil-MC, 2024-03-06:
You need to verify that the log has no missing entries. There is an excellent set of BlackBox Tools written by Jonathan Hudson on Github, which can verify this. He can be contacted on our Discord Forum as @stronnag .

aviphysics, 2024-03-06:
> You need to verify that the log has no missing entries. There is an excellent set of BlackBox Tools written by Jonathan Hudson on Github, which can verify this. He can be contacted on our Discord Forum as @stronnag .

Can you demonstrate that it works with any logs? Why does the BBV5 work with the logs?

Phil-MC, 2024-03-06:
I think BBv5 is probably less fussy and skips over problems. BBv7 might be doing something new. Ask @stronnag on Discord. Sent from my Galaxy

aviphysics, 2024-03-13:
> I think BBv5 is probably less fussy and skips over problems. BBv7 might be doing something new. Ask @stronnag on Discord. Sent from my Galaxy

Do you have any to examples of a successful frequency plot?

Phil-MC, 2024-03-13:
[image]
[image]

These were produced more than a year ago and show vibration before and after replacing a problem motor.

Unfortunately I did not record the version of BB explorer I used but it was 2023/24. But it wasn't the current one!

Phil-MC, 2024-03-13:
What FC generated the BB recording?

rts18, 2024-03-14:
> I don't think that is the problem here (or not the entire problem)
For one thing, BBV5 can render the FFT with scale without a problem from the same dataset that doesn't work in BBV7.

@aviphysics I can confirm your findings. I see infinity on every copter or plane log I record, when viewed with blackbox explorer 7. 
However all logs render the FFT scale consistently with BBE 5.

mmosca, 2024-08-14:
7.0.1 should fix this.

closed 2024-08-14 by mmosca

Phil-MC, 2024-08-14:
Presumably, you refer to the firmware and not the CONFIGURATOR. Does this extend up the line to 7.1.1 and the latest 7.1.2 ?

mmosca, 2024-08-14:
7.0.1 of blackbox log viewer.

Phil-MC, 2024-08-14:
Thanks, I should have thought of that....