iNavFlight/blackbox-log-viewer #95: Workspaces export...  bug of 2019..........................................

Vins1417, 2024-02-28:
Unable to export the workspace, the button exists, but all it does is write null... are there any developers still working on this software? there is an issue from 2019 on this subject ?... how does UAV tech provide us with these files on its Patreon if we can't do it ??

Tonynux, 2024-03-26:
Still export not working!

druckgott, 2024-04-06:
The Function for CustomExport is not Availibe.

If you set the Export with default value in js/main.js then you get an export, but it´s not the current version, it´s a default example.

[image]

If you search for workspaceGraphConfigs you will not find a function. so always item in this area is emptyp 
https://github.com/iNavFlight/blackbox-log-viewer/blob/721ad97f21e9bbf62cdec8031ebc76be379293a3/js/main.js#L796
You will always get this output:
https://github.com/iNavFlight/blackbox-log-viewer/blob/721ad97f21e9bbf62cdec8031ebc76be379293a3/js/main.js#L800

If you search for graphConfig you will finde a function
https://github.com/iNavFlight/blackbox-log-viewer/blob/721ad97f21e9bbf62cdec8031ebc76be379293a3/js/graph_config.js#L4

So anybody have to write the custom function to export this.

druckgott, 2024-04-09:
Fixed it:
https://github.com/iNavFlight/blackbox-log-viewer/pull/99

druckgott, 2024-04-10:
Do anybody have a example for an working input. Because i have some Problems there how the file needs to Look. Like

druckgott, 2024-04-13:
You can also read the log files here: https://blackbox.betaflight.com/ I find out ;-). This should solve your problem.

Tonynux, 2024-04-15:
> Do anybody have a example for an working input. Because i have some Problems there how the file needs to Look. Like

You can ask to Mark Spatz for a working input, I'm sure He has got has one.

Tonynux, 2024-04-15:
> You can also read the log files here: https://blackbox.betaflight.com/ I find out ;-). This should solve your problem. It also reads Inav log files. I did not know it. But its maybe a solution.

I appreciate your willingness to help INAV Users, but we have no problems reading Logs. INAV Blackbox Explorer is OK. We only have problems with Export/Import of workpaces, and... no, unfortunately Betaflight Blackbox Explorer does not solve our problem.

druckgott, 2024-04-15:
Yes i know, but for workspace it is maybe a help of anybody needs this. I did Not know that it works ro read the files there also.