iNavFlight/blackbox-log-viewer #106: Variables in the information page are mixed up.

and-sh, 2024-08-12:
when 
platform = multirotor
pid_type= PIFF

## Current Behavior

Blackbox show nav_fw variables in information page
FC uses nav_mc variables

**fig1.** - all **velosity z** variables set to zero in Inav configurator.

[image]

**fig2.** - althold PID works

[image]

**fig3.** - all **nav_mc_vel_z_*** variables set to zero trough CLI. althold PID does not work.

[image]

## Steps to Reproduce
set 
platform = multirotor
pid_type= PIFF

Try to tune navigation.

## Expected behavior
Show the variables actually used.

## Suggested solution(s)
At least write warning in configurator when such combination are used.

## Additional context

---

Used
INAV 7.1.2 
MATEKF722

mmosca, 2024-08-14:
can you check if it is still broken in 7.0.1?

and-sh, 2024-08-15:
Yes. I need some time