iNavFlight/blackbox-log-viewer #115: Not able to install on Ubuntu 24.04 / Missing dependency

Schildkroet, 2025-03-09:
When trying to install (the latest release) on Ubuntu 24 LTS, i get following error: The following packages have unmet dependencies:
  inav-blackbox-explorer: Hängt ab von: libgconf-2-4 but it is not installable

sensei-hacker, 2025-03-09:
You might try:

wget http://kr.archive.ubuntu.com/ubuntu/pool/universe/g/gconf/libgconf-2-4_3.2.6-6ubuntu1_amd64.deb
wget http://kr.archive.ubuntu.com/ubuntu/pool/universe/g/gconf/gconf2-common_3.2.6-6ubuntu1_all.deb

sudo dpkg -i gconf2-common_3.2.6-6ubuntu1_all.deb
sudo dpkg -i libgconf-2-4_3.2.6-6ubuntu1_amd64.deb

sensei-hacker, 2025-03-10:
The readme (the front page of the repo) also has build instructions. You can try building from source.
I'm still on 22 for the next few weeks,

olexiy-logvinov, 2025-05-13:
I get the same dependency error on Kubuntu 22.04. Building from source does not solve the problem.